Eku127/DualMap #23: Problems in dualmap testing

KingWang569, 2025-09-21:
Dear Dualmap Team

This weekend, I tested a custom scene based on the information we discussed last week. Unfortunately, I encountered some issues:
1. When using the Carter factory in IsaacSim to publish image, depth, and odometry information with a mobile robot, I changed the topic name in the ./config/data_config/ros/self_collected.yaml file to the same topic name as the one used in IsaacSim. However, the terminal kept reporting an error stating that no objects were detected, and neither the local map nor the global map appeared.

2. When I used Record3D on my iPhone for testing, the localmap and globalmap were updated, but I didn't see them saved in the output_path: ./output/map_results directory from the time I started recording until the end of the recording (they can be saved normally using the ROSbag dataset). How can I save the localmap and globalmap of the Record3D stream data? And save it in a data format that can be used for offline query testing

3. I noticed that there is a configuration file for initializing object categories, given_classes_path: ./config/class_list/hm3d300_classes_ycb.txt. How can I customize this file? For example, in some specific field scenarios, such as factories and hospitals, how can I create a txt file corresponding to the indoor environment of a special field?

I look forward to contacting you and your help, thank you ！

[image]

[image]

KingWang569, 2025-09-21:
There are two remaining questions.
4. When using Record3D or IsaacSim, where can I set my navigation goal, e.g., finding my bed?
5. Can I use other sensors (such as Intel's RealSense depth camera) connected to my computer instead of using Record3D for mapping?

Thank you and best wishes！

Eku127, 2025-09-21:
Great to hear that you have the **Record3D** mode running successfully! You can always check the generated **logs** for detailed information.

Regarding your questions:

> However, the terminal kept reporting an error stating that no objects were detected, and neither the local map nor the global map appeared.

This issue could stem from two possible reasons:

1.  **Object Detection Issues**: The detection model might not be identifying objects in your scene.
    * You can verify this by setting `visualize_detection` to `True` in [`config/runner_dataset.yaml#L35`](https://github.com/Eku127/DualMap/blob/e4fa39a6d09e1ea3e9c73ea981ef382a8aea6223/config/runner_dataset.yaml#L35).
    * If you observe that few or no objects are being detected, you can create a custom class list. Simply create a `.txt` file with your class names and place it in the `config/class_list` directory. The system will automatically generate the corresponding `**_colors.json` and `**_id_colors.json` files.

2.  **Point Cloud Processing**: If the system generates bounding boxes on images but still no map observations, the problem likely lies in the point cloud processing.
    * First, please check the `depth_factor`. For the ROS mode, this is currently hard-coded to `1000` in [`applications/utils/runner_ros2.py#L95`](https://github.com/Eku127/DualMap/blob/e4fa39a6d09e1ea3e9c73ea981ef382a8aea6223/applications/utils/runner_ros2.py#L95). This might be the key issue. (I acknowledge this magic number isn't ideal, and I plan to fix it in a future commit).
    * Additionally, you can try to retain more points for generating observations by loosening the point cloud filtering thresholds. These can be adjusted under `PCD Settings` in `config/system_config.yaml`.

---

> How can I save the localmap and globalmap of the Record3D stream data? And save it in a data format that can be used for offline query testing.

To save the maps, you can set `use_end_process` to `True` in [`config/runner_record_3d.yaml#L37`](https://github.com/Eku127/DualMap/blob/e4fa39a6d09e1ea3e9c73ea981ef382a8aea6223/config/runner_record_3d.yaml#L37C1-L37C23). The implementation for this is straightforward and can be found in the `end_process` function here: [`applications/runner_record_3d.py#L145`](https://github.com/Eku127/DualMap/blob/e4fa39a6d09e1ea3e9c73ea981ef382a8aea6223/applications/runner_record_3d.py#L145).

> How can I customize this file? For example, in some specific field scenarios, such as factories and hospitals, how can I create a txt file corresponding to the indoor environment of a special field?

Yes, you can definitely create a custom class list.

As I mentioned before, simply create your list (`.txt` file) and place it in the `config/class_list/` directory. You will also need to update the `class_list` configuration in `config/system_config.yaml` to point to your new file: https://github.com/Eku127/DualMap/blob/e4fa39a6d09e1ea3e9c73ea981ef382a8aea6223/config/system_config.yaml#L25

Eku127, 2025-09-22:
Magic numbers are cleaned and you can pull the latest code and change the depth_factor in the config

closed 2025-11-05 by Eku127