Eku127/DualMap #26: 评估遇到了问题

cybertianli, 2025-09-25:
我在评估单个场景时，先python -m applications.generate_replica_class_color，再python -m evaluation.sem_seg_eval。在预想情况下，我将成功运行评估后看到输出指标，可是，我成功生成
./output/map_results/replica_room_0/classes_info/replica_room_0_names.txt
./output/map_results/replica_room_0/classes_info/replica_room_0_names_colors.json
./output/map_results/replica_room_0/classes_info/replica_room_0_id_colors.json
./output/map_results/replica_room_0/classes_info/replica_room_0_id_names.json
后，运行python -m evaluation.sem_seg_eval时出现了以下错误。`Error executing job with overrides: []
Traceback (most recent call last):
  File "/mnt/8tdisk/wangzhenyu/code/DualMap/evaluation/sem_seg_eval.py", line 683, in main
    evaluator.load_map()
  File "/mnt/8tdisk/wangzhenyu/code/DualMap/evaluation/sem_seg_eval.py", line 387, in load_map
    color = self.given_class_id_colors[obj.class_id]
KeyError: 305
`请问该怎么解决？感谢！

Eku127, 2025-09-25:
你好 @cybertianli ,
问题出在导入的`given_classes_id_color` 这个class list和实际运行生成map所使用的class list不匹配。比如在生产并保存map的时候使用的office的list,但是在评估的时候使用了apartment的list.

你可以首先按照默认的配置进行运行，查看是否可以正常的输出评估的结果，即按照如下的command进行运行：
```
python -m applications.runner_dataset

python -m applications.generate_replica_class_color

given_classes_id_color
```

cybertianli, 2025-09-25:
已成功解决，感谢您的帮助！再次感谢您回复的及时！

closed 2025-09-25 by Eku127